airspy/airspyone_host #16: Strange behavior when reading flash with airspy_spiflash

sapperlott, 2015-01-14:
When writing a firmware image using "airspy_spiflash -w" and re-reading it afterwards using "airspy_spiflash -l 0x1000 -r" the file contents differ. If this isn't a bug, it's not the behavior one would expect after reading the help text of airspy_spiflash.

Using the current v1.0.0-rc3 image (v1.0.0-rc3-0-g4d9e874 2015-01-09) for example, it looks like the read image starts at offset 0x12F0 of the released image.

Another strange thing is that it seems to be impossible to read more than 0x10000 bytes even though the SPI flash chip seems to be a 0x100000 device. When trying to read beyond 0x1000 the AirSpy seems to crash and needs to be replugged to be accessible again.

bvernoux, 2015-01-14:
It is a problem in actual firmware as read in flash does not read anymore the flash, because flash and lot of other peripherals are disabled to lower the internal noise and power consumption.
It is planned (https://github.com/airspy/firmware/wiki/TODO Item 1) to be fixed with new feature in the firmware.
also reading 0x0 to 0x10000 you are just reading the SRAM (because SRAM is remapped to addr 0x0 during startup) and that crash because you are reading something which does not exist after 0x10000.

sapperlott, 2015-01-15:
Thanks for the clarification. Wouldn't it be better then to disable the "-r" option in airspy_spiflash until this issue is fixed to avoid further confusion?

bvernoux, 2015-01-15:
-r option is used for debug/info reading SRAM and peripherals... so no it's not a good idea.

sapperlott, 2015-01-16:
OK, but then I think the documentation should at least reflect the fact that it is reading SRAM instead of SPI flash. Maybe change "(SPIFI@0x80000000)" in the help text to "(SRAM@0x00000000)".

bvernoux, 2015-01-18:
The best change will be probably to remove the help text and add later a wiki dedicated to this tool with full mapping available as read/write ... mainly for developer/tester

touil, 2016-07-26:
The firmware starting from RC9 supports new SPIFI R/W primitives.

closed 2016-07-26 by touil