airspy/airspyone_host #17: airspy_lib_version: symbol lookup error: airspy_lib_version: undefined symbol: airspy_lib_version

itsbert, 2015-01-24:
root@bananapi ~/src/airspy/host (git)-[master] # git clean -dfx
Removing build/
Removing libairspy/build/
root@bananapi ~/src/airspy/host (git)-[master] # git pull
Already up-to-date.
root@bananapi ~/src/airspy/host (git)-[master] # mkdir build
root@bananapi ~/src/airspy/host (git)-[master] # cd build
root@bananapi ~/src/airspy/host/build (git)-[master] # cmake ../ -DINSTALL_UDEV_RULES=ON
-- The C compiler identification is GNU 4.9.1
-- The CXX compiler identification is GNU 4.9.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Found LIBUSB: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Configuring done
-- Generating done
-- Build files have been written to: /root/src/airspy/host/build
cmake ../ -DINSTALL_UDEV_RULES=ON  5.94s user 2.51s system 74% cpu 11.401 total
root@bananapi ~/src/airspy/host/build (git)-[master] # make
Scanning dependencies of target airspy
[  7%] Building C object libairspy/src/CMakeFiles/airspy.dir/airspy.c.o
[ 14%] Building C object libairspy/src/CMakeFiles/airspy.dir/iqconverter_float.c.o
[ 21%] Building C object libairspy/src/CMakeFiles/airspy.dir/iqconverter_int16.c.o
Linking C shared library libairspy.so
[ 21%] Built target airspy
Scanning dependencies of target airspy-static
[ 28%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/airspy.c.o
[ 35%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/iqconverter_float.c.o
[ 42%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/iqconverter_int16.c.o
Linking C static library libairspy.a
[ 42%] Built target airspy-static
Scanning dependencies of target airspy_gpio
[ 50%] Building C object airspy-tools/src/CMakeFiles/airspy_gpio.dir/airspy_gpio.c.o
Linking C executable airspy_gpio
[ 50%] Built target airspy_gpio
Scanning dependencies of target airspy_gpiodir
[ 57%] Building C object airspy-tools/src/CMakeFiles/airspy_gpiodir.dir/airspy_gpiodir.c.o
Linking C executable airspy_gpiodir
[ 57%] Built target airspy_gpiodir
Scanning dependencies of target airspy_info
[ 64%] Building C object airspy-tools/src/CMakeFiles/airspy_info.dir/airspy_info.c.o
Linking C executable airspy_info
[ 64%] Built target airspy_info
Scanning dependencies of target airspy_lib_version
[ 71%] Building C object airspy-tools/src/CMakeFiles/airspy_lib_version.dir/airspy_lib_version.c.o
Linking C executable airspy_lib_version
[ 71%] Built target airspy_lib_version
Scanning dependencies of target airspy_r820t
[ 78%] Building C object airspy-tools/src/CMakeFiles/airspy_r820t.dir/airspy_r820t.c.o
Linking C executable airspy_r820t
[ 78%] Built target airspy_r820t
Scanning dependencies of target airspy_rx
[ 85%] Building C object airspy-tools/src/CMakeFiles/airspy_rx.dir/airspy_rx.c.o
Linking C executable airspy_rx
[ 85%] Built target airspy_rx
Scanning dependencies of target airspy_si5351c
[ 92%] Building C object airspy-tools/src/CMakeFiles/airspy_si5351c.dir/airspy_si5351c.c.o
Linking C executable airspy_si5351c
[ 92%] Built target airspy_si5351c
Scanning dependencies of target airspy_spiflash
[100%] Building C object airspy-tools/src/CMakeFiles/airspy_spiflash.dir/airspy_spiflash.c.o
Linking C executable airspy_spiflash
[100%] Built target airspy_spiflash
make  15.53s user 2.83s system 90% cpu 20.262 total
root@bananapi ~/src/airspy/host/build (git)-[master] # make install
[ 21%] Built target airspy
[ 42%] Built target airspy-static
[ 50%] Built target airspy_gpio
[ 57%] Built target airspy_gpiodir
[ 64%] Built target airspy_info
[ 71%] Built target airspy_lib_version
[ 78%] Built target airspy_r820t
[ 85%] Built target airspy_rx
[ 92%] Built target airspy_si5351c
[100%] Built target airspy_spiflash
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/pkgconfig/libairspy.pc
-- Installing: /usr/local/lib/libairspy.so.1.0.3
-- Up-to-date: /usr/local/lib/libairspy.so.0
-- Up-to-date: /usr/local/lib/libairspy.so
-- Installing: /usr/local/lib/libairspy.a
-- Up-to-date: /usr/local/include/libairspy/airspy.h
-- Up-to-date: /usr/local/include/libairspy/airspy_commands.h
-- Up-to-date: /usr/local/include/libairspy/iqconverter_float.h
-- Up-to-date: /usr/local/include/libairspy/iqconverter_int16.h
-- Up-to-date: /usr/local/include/libairspy/filters.h
-- Up-to-date: /etc/udev/rules.d/52-airspy.rules
-- Installing: /usr/local/bin/airspy_gpio
-- Removed runtime path from "/usr/local/bin/airspy_gpio"
-- Installing: /usr/local/bin/airspy_gpiodir
-- Removed runtime path from "/usr/local/bin/airspy_gpiodir"
-- Installing: /usr/local/bin/airspy_lib_version
-- Removed runtime path from "/usr/local/bin/airspy_lib_version"
-- Installing: /usr/local/bin/airspy_si5351c
-- Removed runtime path from "/usr/local/bin/airspy_si5351c"
-- Installing: /usr/local/bin/airspy_r820t
-- Removed runtime path from "/usr/local/bin/airspy_r820t"
-- Installing: /usr/local/bin/airspy_spiflash
-- Removed runtime path from "/usr/local/bin/airspy_spiflash"
-- Installing: /usr/local/bin/airspy_info
-- Removed runtime path from "/usr/local/bin/airspy_info"
-- Installing: /usr/local/bin/airspy_rx
-- Removed runtime path from "/usr/local/bin/airspy_rx"
root@bananapi ~/src/airspy/host/build (git)-[master] # airspy_lib_version
airspy_lib_version: symbol lookup error: airspy_lib_version: undefined symbol: airspy_lib_version
root@bananapi ~/src/airspy/host/build (git)-[master] # which airspy_lib_version                                                                          :(
/usr/local/bin/airspy_lib_version
root@bananapi ~/src/airspy/host/build (git)-[master] # ldd /usr/local/bin/airspy_lib_version                                                               :(
        libairspy.so.0 => /usr/lib/arm-linux-gnueabihf/libairspy.so.0 (0xb6ee7000)
        libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb6ece000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6eab000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6dbc000)
        /lib/ld-linux-armhf.so.3 (0xb6f18000)
        libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0xb6da2000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6d8c000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6d78000)

itsbert, 2015-01-24:
ran the following and now it works - DOH.
# apt-get remove libairspy-dev libairspy0 airspy
root@bananapi ~/src/airspy/host/build (git)-[master] # airspy_lib_version
AirSpy lib version: 1.0.3

closed 2015-01-24 by itsbert

andimik, 2017-05-08:
Thanks, I had the same problem. Now it works!