airspy/airspyone_host #18: Windows 8.1 airspy_info lists single Aisrpy 32 times.

martinzsmith, 2015-05-23:
Problem description - https://uk.groups.yahoo.com/neo/groups/airspy/conversations/messages/2013
libusb maybe the possible cause of problem ? - https://uk.groups.yahoo.com/neo/groups/airspy/conversations/messages/2030

Can the airspy_open_device function be modified to workaround this Windows 8.1 libusb bug ?
https://github.com/airspy/host/blob/master/libairspy/src/airspy.c#L504

bvernoux, 2015-05-24:
Thanks for this bug report.
It is clearly a bug of libusb (on Win8.1), I prefer to wait anwer from libusb/libusb#56 maintainer before to try a crappy patch for this case.

martinzsmith, 2015-05-24:
Sounds like a very reasonable course of action.

mcuee, 2015-08-26:
I do not see that as a libusb issue (potentially a device issue or driver installation issue) and closed that issue in libusb. You can still reopen the issue if more information are provided to libusb project.

bvernoux, 2015-08-27:
Problem is personally I do not reproduce this problem and the best will be to discuss it with user reproducing this problem

mcuee, 2015-08-28:
No problem. If the user does not come back, then you can always close the ticket. In any case, I do not think this is a libusb issue. It could well be a driver installation issue or that user's machine issue.

bvernoux, 2015-11-23:
Closed this issue cannot be reproduced, can be re-opened if someone find a way to reproduce it.

closed 2015-11-23 by bvernoux