airspy/airspyone_host #27: libusb error

dgorissen, 2015-10-22:
Im running on ubuntu vivid with a fresh clone & install of the airspy repo  (including PR #19 which I understand should make things work out of the box with newer kernels).

However I am unable to connect and unsure how to debug further. See output below.

dgorissen@minion:~$ uname -a
Linux minion 3.19.0-30-generic #34-Ubuntu SMP Fri Oct 2 22:08:41 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

dmesg output:
[ 2000.375749] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[ 2000.610160] usb 1-2: New USB device found, idVendor=1d50, idProduct=60a1
[ 2000.610164] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2000.610165] usb 1-2: Product: AIRSPY
[ 2000.610167] usb 1-2: Manufacturer: www.airspy.com
[ 2000.610168] usb 1-2: SerialNumber: AIRSPY SN:466C64C82C1073C7
[ 2000.641372] airspy 1-2:1.0: usb_control_msg() failed -32 request 0a
[ 2000.641375] airspy 1-2:1.0: Could not detect board
[ 2000.641380] airspy: probe of 1-2:1.0 failed with error -32

dgorissen@minion:~$ airspy_info 
airspy_lib_version: 1.0.6

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc5-0-g648c14f 2015-05-20
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x466C64C82C1073C7
Supported sample rates:
	10.000000 MSPS
	2.500000 MSPS
Close board 1
airspy_close() board 1 failed: AIRSPY_ERROR_LIBUSB (-1000)

dgorissen, 2015-10-22:
PS: just to add that it works fine with the windows VM I have on the same machine. So its definitely not a hardware issue.  Have also updated the firmware to rc6 but no improvement (if anything its worse as airspy_info fails directly with the libusb error without printing any info first)

martinzsmith, 2015-10-22:
Your kernel version is greater than 3.17 have you blacklisted the Linux kernel Airspy driver ?
See "Kernel >= 3.17 problems" https://github.com/airspy/host/wiki/Troubleshooting#linux-debianubuntu

dgorissen, 2015-10-22:
Blacklisting has seemed to solve it. Doh! Thanks.

martinzsmith, 2015-10-22:
Sorry I did not realised that the "libusb_detach_kernel_driver for Linux Kernel" was added to the master branch.

bvernoux, 2015-10-22:
I have updated libairspy version to 1.0.7 in order to be sure we use the feature "libusb_detach_kernel_driver for Linux Kernel" and also other feature introduced after 1.0.6
Could you try again without blacklisting kernel module ?
As this version shall fix the problem.

dgorissen, 2015-10-22:
Unfortunately it does not seem fixed (enabling the blacklist fixes things again)

dgorissen@minion:~/host.orig/build$ airspy_info 
airspy_lib_version: 1.0.7

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc6-0-g035ff81 2015-07-14
airspy_board_partid_serialno_read() failed: AIRSPY_ERROR_LIBUSB (-1000)

bvernoux, 2015-10-23:
Could you try with sudo airspy_info ?

bvernoux, 2015-10-24:
Could you apply this patch
https://gist.github.com/bvernoux/7175fad615c290391ffb
On latest source code https://github.com/airspy/host/commit/4a1c734d691486bd22ada4d741e543683ab4458a
Then send me the log when calling airspy_info
And also tell me if that fix the problem

dgorissen, 2015-10-24:
Running as root makes no difference. Applied patch, log below.

dgorissen@minion:~/host.orig/build$ airspy_info 
airspy_lib_version: 1.0.7
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000680] [00005696] libusb: debug [libusb_kernel_driver_active] interface 0
libusb_kernel_driver_active() res=0
[ 0.011624] [00005696] libusb: debug [libusb_detach_kernel_driver] interface 0
libusb_detach_kernel_driver() res=-5
[ 0.011681] [00005696] libusb: debug [libusb_set_configuration] configuration 1
[ 0.011754] [00005697] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.050202] [00005696] libusb: debug [libusb_claim_interface] interface 0
[ 0.050370] [00005696] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.050465] [00005696] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.050713] [00005696] libusb: debug [handle_events] poll() returned 1
[ 0.050769] [00005696] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=1
[ 0.050795] [00005696] libusb: debug [handle_control_completion] handling completion status 0
[ 0.050819] [00005696] libusb: debug [disarm_timerfd] 
[ 0.050831] [00005696] libusb: debug [usbi_handle_transfer_completion] transfer 0x111b110 has callback 0x7fcba0e130b0
[ 0.050854] [00005696] libusb: debug [sync_transfer_cb] actual_length=1
[ 0.051803] [00005696] libusb: debug [libusb_init] libusb v1.0.19.10903
[ 0.051940] [00005696] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.051981] [00005696] libusb: debug [op_init] bulk continuation flag supported
[ 0.052017] [00005696] libusb: debug [op_init] zero length packet flag supported
[ 0.052062] [00005696] libusb: debug [op_init] sysfs can relate devices
[ 0.052098] [00005696] libusb: debug [op_init] sysfs has complete descriptors
[ 0.052345] [00005696] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.057135] [00005696] libusb: debug [linux_get_device_address] scan usb1
[ 0.057323] [00005696] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.057410] [00005696] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.057477] [00005696] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.057779] [00005696] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.057865] [00005696] libusb: debug [linux_get_device_address] scan 1-1
[ 0.057973] [00005696] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.058071] [00005696] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.058179] [00005696] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.058419] [00005696] libusb: debug [linux_get_parent_info] Dev 0x111e160 (1-1) has parent 0x111e0a0 (usb1) port 1
[ 0.066897] [00005696] libusb: debug [linux_get_device_address] getting address for device: 1-2 detached: 0
[ 0.066910] [00005696] libusb: debug [linux_get_device_address] scan 1-2
[ 0.067019] [00005696] libusb: debug [linux_get_device_address] bus=1 dev=6
[ 0.067198] [00005696] libusb: debug [linux_enumerate_device] busnum 1 devaddr 6 session_id 262
[ 0.067210] [00005696] libusb: debug [linux_enumerate_device] allocating new device for 1/6 (session 262)
[ 0.067285] [00005696] libusb: debug [linux_get_parent_info] Dev 0x111e220 (1-2) has parent 0x111e0a0 (usb1) port 2
[ 0.067385] [00005696] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.067398] [00005696] libusb: debug [linux_get_device_address] scan usb2
[ 0.067452] [00005696] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.067464] [00005696] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.067472] [00005696] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.067567] [00005696] libusb: debug [usbi_add_pollfd] add fd 12 events 1
[ 0.067581] [00005696] libusb: debug [usbi_add_pollfd] add fd 14 events 1
[ 0.067590] [00005696] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.067596] [00005696] libusb: debug [usbi_add_pollfd] add fd 16 events 1
[ 0.067604] [00005696] libusb: debug [libusb_get_device_list] 
[ 0.067616] [00005696] libusb: debug [libusb_get_device_descriptor] 
[ 0.067623] [00005696] libusb: debug [libusb_get_device_descriptor] 
[ 0.067629] [00005696] libusb: debug [libusb_open] open 1.6
[ 0.067650] [00005696] libusb: debug [usbi_add_pollfd] add fd 17 events 4
[ 0.067661] [00005696] libusb: debug [libusb_kernel_driver_active] interface 0
libusb_kernel_driver_active() res=0
[ 0.067676] [00005696] libusb: debug [libusb_detach_kernel_driver] interface 0
libusb_detach_kernel_driver() res=-5
[ 0.067689] [00005696] libusb: debug [libusb_set_configuration] configuration 1
[ 0.067709] [00005696] libusb: debug [libusb_close] 
[ 0.067718] [00005696] libusb: debug [usbi_remove_pollfd] remove fd 17
[ 0.067728] [00005696] libusb: debug [libusb_get_device_descriptor] 
[ 0.067735] [00005696] libusb: debug [libusb_get_device_descriptor] 
[ 0.067743] [00005696] libusb: debug [libusb_exit] 
[ 0.067750] [00005696] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.067757] [00005696] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[... 13 lines cut ...]
[... cut ...]
[ 6.689135] [00005696] libusb: debug [handle_control_completion] handling completion status -32
[ 6.689141] [00005696] libusb: debug [handle_control_completion] unsupported control request
[ 6.689146] [00005696] libusb: debug [disarm_timerfd] 
[ 6.689152] [00005696] libusb: debug [usbi_handle_transfer_completion] transfer 0x111ed90 has callback 0x7fcba0e130b0
[ 6.689156] [00005696] libusb: debug [sync_transfer_cb] actual_length=0
airspy_board_partid_serialno_read() failed: AIRSPY_ERROR_LIBUSB (-1000)

bvernoux, 2015-10-25:
Thanks for the feedback so the problem is confirmed
libusb_kernel_driver_active() res=0 => No airspy kernel driver detected why ?
libusb_detach_kernel_driver() res=-5 => LIBUSB_ERROR_NOT_FOUND = -5 (probably because No airspy kernel driver detected)
For me it seems to be a bug with libusb with your kernel airspy module loaded which is not detected so it cannot be unloaded.
We will need to check with libusb developer why such error happen with your system

bvernoux, 2015-10-26:
Thanks to Antti for his quick fix see
http://git.linuxtv.org/cgit.cgi/anttip/media_tree.git/commit/?h=airspy&id=a4db80b38a537bb58b73a634238d7519c0f57e2a
So old driver and actual ones (until version is compliant with AirSpy v1.0) will do not match version string required by kernel driver and so and the kernel driver will exit and so we will do not need to blacklist or unload the airspy kernel driver.

bvernoux, 2015-12-23:
So this problem shall be fixed with linux kernel [v4.4-rc6](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/log/?id=refs/tags/v4.4-rc6)

closed 2016-10-24 by bvernoux