airspy/airspyone_host #28: [1423] closing error

bibib86, 2015-10-25:
When I close the program, I get this error.
The filter audio plugin is not loaded.
---------------------------------------------------------------------------------------------------------------------------
Consultez la fin de ce message pour plus de détails sur l'appel du débogage
juste-à-temps (JIT) à la place de cette boîte de dialogue.

************** Texte de l'exception **************
System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à SDRSharp.MainForm.MainForm_Closing(Object sender, CancelEventArgs e)
   à System.Windows.Forms.Form.OnClosing(CancelEventArgs e)
   à System.Windows.Forms.Form.WmClose(Message& m)
   à System.Windows.Forms.Form.WndProc(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   à System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   à System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Assemblys chargés **************
mscorlib
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34209 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
SDRSharp
    Version de l'assembly : 1.0.0.1423
    Version Win32 : 1.0.0.1423
    CodeBase : file:///C:/Users/Administrateur/Desktop/SDR/sdr-install/sdrsharp/SDRSharp.exe
----------------------------------------
System
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34238 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
SDRSharp.Radio
    Version de l'assembly : 0.0.0.0
    Version Win32 : 0.0.0.0
    CodeBase : file:///C:/Users/Administrateur/Desktop/SDR/sdr-install/sdrsharp/SDRSharp.Radio.DLL
----------------------------------------
System.Windows.Forms
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34251 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34270 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
SDRSharp.Common
    Version de l'assembly : 0.0.0.0
    Version Win32 : 0.0.0.0
    CodeBase : file:///C:/Users/Administrateur/Desktop/SDR/sdr-install/sdrsharp/SDRSharp.Common.DLL
----------------------------------------
SDRSharp.PanView
    Version de l'assembly : 0.0.0.0
    Version Win32 : 0.0.0.0
    CodeBase : file:///C:/Users/Administrateur/Desktop/SDR/sdr-install/sdrsharp/SDRSharp.PanView.DLL
----------------------------------------
System.Configuration
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34209 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34234 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
SDRSharp.CollapsiblePanel
    Version de l'assembly : 0.0.0.0
    Version Win32 : 0.0.0.0
[... 149 lines cut ...]
    Version Win32 : 4.0.30319.34234 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.0.30319.34209 built by: FX452RTMGDR
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** Débogage JIT **************
Pour activer le débogage juste-à-temps (JIT), le fichier de configuration pour cette
application ou cet ordinateur (machine.config) doit avoir la valeur
jitDebugging définie dans la section system.windows.forms.
L'application doit également être compilée avec le débogage
activé.

Par exemple :

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Lorsque le débogage juste-à-temps est activé, les exceptions non gérées
seront envoyées au débogueur JIT inscrit sur l'ordinateur
plutôt que d'être gérées par cette boîte de dialogue.

touil, 2015-10-25:
This has nothing to do with libairspy.

closed 2015-10-25 by touil

bibib86, 2015-10-26:
Ok....
can be a little help?
since the forum is closed, I do not know where to go ...