airspy/airspyone_host #29: Bug in airspy_set_samplerate()

Nicolasp38, 2015-11-23:
Hi,

Just reading source code for my knowledge.In file airspy.c, function airspy_set_samplerate(), line 882, I believe there is a bug : LIBUSB_ENDPOINT_IN should in fact be LIBUSB_ENDPOINT_OUT.
Right ?

Nicolas

bvernoux, 2015-11-23:
There is no any problem with this function LIBUSB_ENDPOINT_IN is ok and the parameters are ok and all work like expected (tested with Windows XP, Seven, Win10 ..., Linux, MacOS).

closed 2015-11-23 by bvernoux

Nicolasp38, 2015-11-24:
If I understand correctly, airspy_set_samplerate() tell the USB device to set its sample rate to a defined state, so this is a host-to-device transaction. Host-to-device must use LIBUSB_ENDPOINT_OUT, not LIBUSB_ENDPOINT_IN.

bvernoux, 2015-11-24:
It is not used like that to avoid doing a DATA stage for nothing (and all is passed during SETUP stage instead of having an other stage with DATA)

Nicolasp38, 2015-11-25:
Ok. Thanks for the explanation.