airspy/airspyone_host #31: patches for visual studio

guruofquality, 2015-12-06:
Hi all,

I'm building airspy latest release (v1.0.7) for the PothosSDR windows installer. We include airspy support for SoapySDR and GrOsmoSDR: More info here: https://github.com/pothosware/PothosSDR/wiki

I had to make few changes to get the build working on MSVC 2012 and 2013. This is the patch to get everything building, and I will break it down below: https://github.com/pothosware/PothosSDR/blob/master/patches/airspy_build_msvc.diff

custom pthread and libusb library paths
====

I provide a libusb and pthread library for the environment. And I pass in pthread and libusb to many of the libraries, not just airspy. I needed a way to disable the following code when custom pthread and libusb library paths are specified. I would be happy for an alternative mechanism though. Also, I could not set CMAKE_CROSSCOMPILING=ON as a work-around hack. CMake will override that setting.
```
 if(WIN32 AND NOT CMAKE_CROSSCOMPILING)
+ if (NOT DEFINED LIBUSB_LIBRARIES)
  configure_file(
      ${CMAKE_CURRENT_BINARY_DIR}/../libs_win32/libusb-1.0.dll
      ${CMAKE_CURRENT_BINARY_DIR}/airspy-tools/src/libusb-1.0.dll
  COPYONLY)
+ endif()
 
+ if (NOT DEFINED THREADS_PTHREADS_WIN32_LIBRARY)
  configure_file(
      ${CMAKE_CURRENT_BINARY_DIR}/../libs_win32/pthreadGC2.dll
      ${CMAKE_CURRENT_BINARY_DIR}/airspy-tools/src/pthreadGC2.dll
  COPYONLY)
+ endif()
 endif()
```

building with vc11 (msvc 2012)
====

I build for both msvc 2012 and 2013 targets (eventually 2015). I discovered some minor build issues for 2012 that were easy to remedy. Here are the patches.

The strtoull() function was missing in both the library and tools. The simplest work around for me was to define in the top level with a macro, since _strtoui64() provides the same functionality:
```
+#provide missing strtoull() for VC11
+if(MSVC11)
+    add_definitions(-Dstrtoull=_strtoui64)
+endif(MSVC11)
```

The IQ converter float code actually had a version check in it, but the code must have changed around it. Since I was able to build for vc12 but not vc11, I updated the check for ```_MSC_VER >= 1800``` which corresponds to vc12 and up. The converter code now builds for both compiers with this new ifdef. I assume that this change means that the implementation uses a non-accelerated converter for vc11:

```
   //#define FIR_AUTO_VECTOR
 #else
-	#if (_MSC_VER >= 1300)
+	#if (_MSC_VER >= 1800)
 		#define USE_SSE2
 		#include <immintrin.h>
 	#else
```

Thanks for your consideration!

bvernoux, 2015-12-06:
Thanks for the details
Could you add a Pull Request on dev branch in order to test your patch ?

guruofquality, 2015-12-06:
Here you are. I split the patch into 3 commits so it could be cherry-picked if need-be: https://github.com/airspy/host/pull/32

bvernoux, 2015-12-06:
Very nice work !!
I have merged it in dev branch for test purpose and to confirm all work fine before to merge it in master

bvernoux, 2015-12-23:
Thanks for the patch now it is merged in master see Dev #34

closed 2015-12-23 by bvernoux

guruofquality, 2015-12-24:
thanks, appreciated!