airspy/airspyone_host #33: New functions: airspy_set_linearity_gain and airspy_set_sensitivity_gain

amontefusco, 2015-12-12:
Hi all,

I would like to implement in my ExtIO dll two new sliders, in order to use the new functions.

So I created two slider with 0-21 range, one acting on airspy_set_linearity_gain, the other one on airspy_set_sensitivity_gain.

However it is not clear to me the logic of those functions: it seems that one overwrite the parameters of other.
So, changing the slider for sensitivity make the position of linearity slider inconsistent and viceversa. 

Could you suggest a possible working arrangement for that two sliders ?

touil, 2015-12-12:
There are three gain "modes" and they are mutually exclusive. I propose having radio buttons to switch between the "Linearity", "Sensitivity" and "Custom" modes like in SDR#.

amontefusco, 2015-12-12:
Ok for the three modes.

There is even another problem, maybe minor: flipping (as an example) from sensitivity mode back to manual, it is unfeasible to set properly the sliders for each gain value, as there is no function in the API to retrieve the current gains values.
So the sliders for the three gains would be in an inconsistent status coming back to manual from "automatic" .
For the time being, I just restore the previously saved "manual" status.

touil, 2015-12-12:
Restoring the old manual values should be fine. That's what I do in SDR# anyway.

closed 2015-12-12 by amontefusco