airspy/airspyone_host #35: FreeBSD support

akhenakh, 2016-01-04:
I've done some experiment to make airspy works on FreeBSD (rasperry).
Seems to be as simple as this patch but cmakes

```
diff --git a/libairspy/src/iqconverter_float.c b/libairspy/src/iqconverter_float.c
index 844e846..111007e 100644
--- a/libairspy/src/iqconverter_float.c
+++ b/libairspy/src/iqconverter_float.c
@@ -38,6 +38,12 @@ THE SOFTWARE.
   #define _aligned_free(mem) free(mem)
   #define _inline inline
   #define FIR_STANDARD
+#elif defined(__FreeBSD__)
+  #include <stdlib.h>
+  #define _aligned_malloc(size, alignment) malloc(size)
+  #define _aligned_free(mem) free(mem)
+  #define _inline inline
+  #define FIR_STANDARD
 #elif defined(__GNUC__) && !defined(__MINGW64_VERSION_MAJOR)
   #include <malloc.h>
   #define _aligned_malloc(size, alignment) memalign(alignment, size)
diff --git a/libairspy/src/iqconverter_int16.c b/libairspy/src/iqconverter_int16.c
index d38df68..50798db 100644
--- a/libairspy/src/iqconverter_int16.c
+++ b/libairspy/src/iqconverter_int16.c
@@ -34,6 +34,11 @@ THE SOFTWARE.
   #define _aligned_malloc(size, alignment) malloc(size)
   #define _aligned_free(mem) free(mem)
   #define _inline inline
+#elif defined(__FreeBSD__)
+  #include <stdlib.h>
+  #define _aligned_malloc(size, alignment) malloc(size)
+  #define _aligned_free(mem) free(mem)
+  #define _inline inline
 #elif defined(__GNUC__) && !defined(__MINGW64_VERSION_MAJOR)
   #include <malloc.h>
   #define _aligned_malloc(size, alignment) memalign(alignment, size)
```

Using cmake 3.4.1 (the one available in FreeBSD ports) it seems you need to add set(THREADS_PREFER_PTHREAD_FLAG ON) to correctly detect the right libpthread (without s) but still cmake is running its tests with -lpthreads, I don't know cmake enough to fix this.

The manually compiled binaries seems to work but airpsy_info detects the same board 32 times:
```
Found AirSpy board 32
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc3-0-g4d9e874 2015-01-09
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x16C46XXXXX
Supported sample rates:
        10.000000 MSPS
        2.500000 MSPS
Close board 32
```
Any ideas? Thanks.

bvernoux, 2016-01-04:
Could you test with latest firmware rc6 ?
This problem with 32 boards found seems related to a bug in an old version of libusb to be confirmed versus Ubuntu 14.04 LTS which works fine

akhenakh, 2016-01-04:
Same issue with latest firmware :/

I'll try to investigate a bit further.
```
Found AirSpy board 32
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc6-0-g035ff81 2015-07-14
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x16C4XXXXXx
Supported sample rates:
	10.000000 MSPS
	2.500000 MSPS
Close board 32
```

bvernoux, 2016-01-04:
It will be interesting to find the root cause of this issue even if it is not blocker as we can use correctly airspy the bad effect is I doubt multiple board work fine with this bug.
Could you test airspy_rx -r /dev/null -t 0 to see if it work fine ?
and also with -t 1, 2, 3 and 4 running during at least 30s

akhenakh, 2016-01-05:
It doesn't seem to act properly, 

On Linux same hardware, it goes around 3MSPS with no linearity.
Probably here it doesn't receive any signal.
```
Streaming at 10.00 MSPS
Streaming at 10.00 MSPS
Streaming at 10.00 MSPS
Streaming at  8.10 MSPS
Streaming at  8.10 MSPS
Streaming at  8.10 MSPS
Streaming at  8.10 MSPS
Streaming at  8.10 MSPS
Streaming at  8.10 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  6.58 MSPS
Streaming at  5.37 MSPS
Streaming at  5.37 MSPS
Streaming at  5.37 MSPS
Streaming at  5.37 MSPS
Streaming at  5.37 MSPS
Streaming at  5.37 MSPS
Streaming at  4.40 MSPS
Streaming at  4.40 MSPS
Streaming at  4.40 MSPS
Streaming at  4.40 MSPS
Streaming at  4.40 MSPS
Streaming at  4.40 MSPS
Streaming at  3.62 MSPS
Streaming at  3.62 MSPS
q^CCaught signal 2

User cancel, exiting...
Total time: 34.1621 s
Average speed 5.6163 MSPS IQ
```

bvernoux, 2016-01-05:
Could you check on same computer with Ubuntu 14.04 to check if all work fine ? (as you could have issue with your computer too)

martinzsmith, 2016-07-26:
>  I've done some experiment to make airspy works on FreeBSD (**rasperry**).

@akhenakh  The Raspberry Pi is probably the worst hardware that you could possibly choose to try and debug problems on. It's USB port performance is sub-optimal, especially if you are connecting to it over the network as it is using the same, one and only real, USB port for that as well. If you were testing at 2.5MSPS it might possibly be OK or if you are using something like the following to test "airspy_rx -r /dev/null -p 1" 

If I was trying to write patches to get the Airspy to work under *BSD,  I'd probably start by looking at what was done to get the HackRF One working. 
http://svnweb.freebsd.org/ports/head/comms/hackrf/
The original code base for the Airspy was derived from the HackRF code, but most things have changed a lot. But if someone is able to get the HackRF working under *BSD by modifying two Cmake files, then it should be a similar level of difficulty to getting the Airspy library and host tools working natively under *BSD.

akhenakh, 2016-07-26:
I haven't tried it again recently.
Same raspberry pi hw (at least 2 if I remember correctly) was capable of handling the 10MSPS of the airspy, so BSD* on same hw should probably do it too.
thanks for the infos I may give it another shot some days!

bvernoux, 2017-04-04:
Do you have any update about that ?

bvernoux, 2018-01-06:
Added Support FreeBSD #50 thanks to @Unitrunker contribution
Any feedback is welcome

closed 2018-01-06 by bvernoux