airspy/airspyone_host #36: CMake Error at /usr/share/cmake/Modules/TestBigEndian.cmake with GCC 4.4

dranch, 2016-02-07:
When compiling the Airspu host libraries and tools on Centos6.7, running the "cmake ../ -DINSTALL_UDEV_RULES=ON" command gives the following error:

-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - not found
-- Looking for stddef.h
-- Looking for stddef.h - not found
-- Check size of unsigned short
-- Check size of unsigned short - failed
-- Check size of unsigned int
-- Check size of unsigned int - failed
-- Check size of unsigned long
-- Check size of unsigned long - failed
CMake Error at /usr/share/cmake/Modules/TestBigEndian.cmake:44 (message):
  no suitable type found
Call Stack (most recent call first):
  libairspy/CMakeLists.txt:73 (TEST_BIG_ENDIAN)

Per this HackRF thread ( https://pairlist9.pair.net/pipermail/hackrf-dev/2013-July/000120.html ), the libairspy/CMakeLists.txt and airspy-tools/CMakeLists.txt files improperly specify the gnu90 ANSI standard but GCC v4.4 only supports GNU89.  Simply deleting these two lines fixes the issue and things compile fine.

--David

bvernoux, 2016-02-08:
Thanks for the comment
Could you do a pull request with the fixes (IIRC 2 files are concerned by this option gnu90) ? (like that you will benefit of this)

bvernoux, 2016-02-21:
Could you confirm your problem is fixed ? (in order to close this issue)

closed 2016-04-23 by bvernoux