airspy/airspyone_host #39: How to handle more than one device ?

f4gkr, 2016-10-08:
Hi AirSpy team,

I am trying to manage more than one device connected on the same computer. Assuming I do not know the serial numbers (hence cannot call airspy_open_sn() ) is there a way to retrieve the list of struct airspy_device* attached ?

Merci
best from F4GKR Sylvain

bvernoux, 2016-10-09:
The other way to open multiple device without knowing serial number is to call airspy_open() multiple times (without closing the device)
and it will open 1st board found then 2nd board found and so on ... exactly like it is done with [airspy_info.c](https://github.com/airspy/host/blob/master/airspy-tools/src/airspy_info.c#L135) when you do not provide a serial number

closed 2016-10-09 by bvernoux

f4gkr, 2016-10-09:
Merci

sylvain