airspy/airspyone_host #44: FindUSB1.cmake error under RPI3

andimik, 2017-05-04:
I have several PCs where I could install the host under Ubuntu. But on my RPI3 I get an error:

> pi@raspberrypi:~/apps/airspy/airspyone_host-master/libairspy/build $ cmake ../ -DINSTALL_UDEV_RULES=ON
> CMake Error at CMakeLists.txt:77 (find_package):
>   By not providing "FindUSB1.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "USB1", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "USB1" with any of
>   the following names:
> 
>     USB1Config.cmake
>     usb1-config.cmake
> 
>   Add the installation prefix of "USB1" to CMAKE_PREFIX_PATH or set
>   "USB1_DIR" to a directory containing one of the above files.  If "USB1"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/pi/apps/airspy/airspyone_host-master/libairspy/build/CMakeFiles/CMakeOutput.log".
> 

What can I do?

> pi@raspberrypi:~ $ dpkg -l | grep usb
> ii  libusb-0.1-4:armhf                     2:0.1.12-25                               armhf        userspace USB programming library
> ii  libusb-1.0-0:armhf                     2:1.0.19-1                                armhf        userspace USB programming library
> ii  libusb-1.0-0-dbg:armhf                 2:1.0.19-1                                armhf        userspace USB programming library development files
> ii  libusb-1.0-0-dev:armhf                 2:1.0.19-1                                armhf        userspace USB programming library development files
> ii  libusb-1.0-doc                         2:1.0.19-1                                all          documentation for userspace USB programming
> ii  libusbmuxd2:armhf                      1.0.9-1                                   armhf        USB multiplexor daemon for iPhone and iPod Touch devices - library
> ii  usb-modeswitch                         2.2.0+repack0-2                           armhf        mode switching tool for controlling "flip flop" USB devices
> ii  usb-modeswitch-data                    20150115-1                                all          mode switching data for usb-modeswitch
> ii  usbutils                               1:007-2                                   armhf        Linux USB utilities
>

bvernoux, 2017-05-04:
It seems you have not executed the **Prerequisites for Linux (Debian/Ubuntu/Raspbian)** see
https://github.com/airspy/airspyone_host#how-to-build-the-host-software-on-linux mainly step 

> sudo apt-get install build-essential cmake libusb-1.0-0-dev pkg-config

Please do all those steps and you shall do not have any problem as your problem seems related to the fact libusb headers/library are not found

andimik, 2017-05-04:
No, I've installed them, they are the latest version.

> pi@raspberrypi:/ $ sudo apt-get install build-essential cmake libusb-1.0-0-dev pkg-config
> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut.       
> Statusinformationen werden eingelesen.... Fertig
> build-essential ist schon die neueste Version.
> libusb-1.0-0-dev ist schon die neueste Version.
> pkg-config ist schon die neueste Version.
> cmake ist schon die neueste Version.
> Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
>   libqwt6
> Verwenden Sie »apt-get autoremove«, um es zu entfernen.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 20 nicht aktualisiert.
>

bvernoux, 2017-05-04:
I have checked your path and it is wrong
**pi@raspberrypi:~/apps/airspy/airspyone_host-master/libairspy/build $ cmake ../ -DINSTALL_UDEV_RULES=ON**
It shall be
**pi@raspberrypi:~/apps/airspy/airspyone_host-master/build $ cmake ../ -DINSTALL_UDEV_RULES=ON**
You have not created the build directory in the right path so it cannot work to find modules ...
Please do exactly like it is described here **https://github.com/airspy/airspyone_host#how-to-build-the-host-software-on-linux** and do not change the path for the build dir
remove all files previously downloaded... and retry all step by step

touil, 2017-05-04:
RTFM.

closed 2017-05-04 by touil

andimik, 2017-05-04:
Very strange. I used wget and this now works. I always used the green button for downloading the zip-file.