airspy/airspyone_host #45: Optional filter

nbonniere, 2017-07-01:
I would like to use an Airspy-mini for the Open-Glider-Network at 3 MHz sampling and decimate by 3 for 1 MHz bandwidth and specify an appropriate filter with:

extern ADDAPI int ADDCALL airspy_set_conversion_filter_float32(struct airspy_device* device, const float *kernel, const uint32_t len);

I have tried various filter designs, but they all end up with aliasing. Can you provide details for the design of the required filter taps ?

touil, 2017-07-01:
You are probably confusing the decimation filters with the conversion filters used to convert the raw (real) signal out of the ADC to complex analytic signal (IQ).
FYI, any extra DSP work not included in user mode driver (libairspy) must be performed by the client application (ie. your program).
If you have a serious use case with critical real-time and reliability constraints, we can do custom code for you at expense.

closed 2017-07-01 by touil

nbonniere, 2017-07-01:
As I said, I need 1 MHz sampling to provide 1 MHz bandwidth. This is easy with an RTLSDR dongle.  I was trying to use an Airspy-mini with more than 8 bits to see if I can get a better performing OGN receiver (http://wiki.glidernet.org/). I know that the conversion filter is not done on the Airspy (unfortunately, not enough CPU in the processor ?) and uses CPU which is at a premium on a Raspberry Pi-3. To avoid doing the extra filtering (lots of CPU) prior to decimation, I was trying to narrow the conversion filter to 1/3 and then decimate by 3. I was successful in narrowing the filter, but the resulting spectrum was aliased. The Airspy IF is set to 1/4 the sampling rate, so it is easy to convert the samples to IQ (SIN/COS 0, 90, 180, 270), and I was looking for a way to change the filter taps to avoid further processing on the RPi. In fact with 3x decimation you can avoid calculating 2/3 of the samples and use the CPU for more taps for a narrower filter.

touil, 2017-07-01:
Again, you are probably trying to fix a problem you created by yourself by not choosing the right design options.
The issue tracker is not the best place to discuss requirements specific to your project. However, if you find a bug in the library, we would be more than glad to fix it.

nbonniere, 2017-07-01:
You are right. I should not have chosen an Airspy because it has very limited sampling capability, and basically only does ADC and pushes all the processing onto the user instead of doing it internally.  Thank you for your un-helpfulness.  SDR# is a very nice program, but if you want to do more than look at a spectrum and actually do GFSK demodulation you are out of luck.

touil, 2017-07-01:
You are free to learn or not learn how to fix your problems. Same as for learning how to gently ask someone else who can fix your own problem.

nbonniere, 2017-07-01:
I was hoping that you could tell me how to use the API  airspy_set_conversion_filter_float32 function to provide a narrower filter which would then allow for decimation. I guess that is not possible.  I'll have to use raw mode and do the FIR filtering in my own code, with decimation. The RPI-3 doesn't have a lot of horsepower and I didn't want to have two FIR filters using CPU.

touil, 2017-07-01:
Again, airspy_set_conversion_filter_float32 does not apply a filter on the IQ stream. It cannot be used for decimation and has completely different purpose. Also, the RAW mode doesn't even give you IQ signals. It only gives you the In-Phase signal at double the sample rate.
If you just want to decimate by 3, then use one of the available IQ modes and add a decimating FIR externally. You can use a GNU Radio block for that.

nbonniere, 2017-07-01:
Yes, I know raw mode is not IQ, and yes it is easy to add an FIR filter and do decimation, but I wanted to avoid lots of extra CPU that the FIR requires and that is simply not available on an RPI-3. The RPi OGN rtlsdr receiver code can take 1 or 2 MHz sampling not 6 or 3, so decimation by 3 is needed. When I saw the conversion filter I thought there might be a way to combine the conversion filter and the FIR for decimation. Perhaps in the future you can consider adding a decimation API function,

touil, 2017-07-01:
You can use Parks-McClellan to produce a half-band with 19 taps and 0.3*w pass band as conversion filter; and another one with 79 taps, 0.3*w pass band and 0.366667*w stop band for decimation by 3.

You are supposed to do this, not the device driver.

nbonniere, 2017-07-02:
Thank you. That is very helpful. What I didn't realize was that every second tap is skipped/not-calculated in the conversion filter code and I was providing airspy_set_conversion_filter_float32 with an FIR filter that had non-zero taps.

touil, 2017-07-02:
Note that my suggestion assumes you need 90% of the spectrum alias-free. If you need less, you can save even more cpu time.