airspy/airspyone_host #46: Can three additional functions be added to libairspy

martinzsmith, 2017-12-12:
airspy_get_lna_gain
airspy_get_mixer_gain
airspy_get_vga_gain
ref: https://github.com/pothosware/SoapyAirspy/pull/14#issuecomment-351199121

bvernoux, 2017-12-12:
I do not understand such requirement as such value comes from the user which call the airspy_set_xxx so it shall be saved locally .and reused later without the need to add such API in libairspy

martinzsmith, 2017-12-13:
The problem is if you call airspy_set_linearity_gain (airspy_set_sensitivity_gain) it modifies the lna, mixer and vga gains, now the client software is keeping track of them them also and they have been modified and it has no idea what they currently are. A different workaround would be to fully duplicate this lookup table:
https://github.com/airspy/airspyone_host/blob/master/libairspy/src/airspy.c#L117

in SoapyAirspy.

bvernoux, 2017-12-13:
Thanks for the clarification I have not checked that aspect in that case we could add such APIs in libairspy any contribution is welcome (but it shall not require to modify the firmware)

bvernoux, 2022-02-25:
This issue is old and I have no time to work on it, so it is closed
If anyone is interested you are welcome to do a Pull Request

closed 2022-02-25 by bvernoux