airspy/airspyone_host #56: License file

nowls, 2018-04-10:
Could you please add a LICENSE file in the repository root directory as you did with libairspyhf?

bvernoux, 2018-04-10:
It is not as simple as there is multiple licenses depending on directories:

- Files under https://github.com/airspy/airspyone_host/tree/master/airspy-tools are [GPL-2.0](https://opensource.org/licenses/GPL-2.0)
- Files under https://github.com/airspy/airspyone_host/tree/master/libairspy are [BSD-3-Clause](https://opensource.org/licenses/BSD-3-Clause)

So we could probably add that in each directory to clarify the license for each part and maybe add this comment at end of https://github.com/airspy/airspyone_host/blob/master/README.md

Any proposal are welcome

nowls, 2018-04-10:
I think the license-file-per-component model works well. The reason I'm requesting this feature is so I can point to a license file to build the `libairspy` component in `OpenEmbedded`. The build system tracks license files for you so embedded Linux developers can easily verify license compliance throughout the build.

bvernoux, 2018-04-10:
I have added license for [airspy-tools](https://github.com/airspy/airspyone_host/blob/master/airspy-tools/LICENSE.md) and [airspy-tools/getopt](https://github.com/airspy/airspyone_host/blob/master/airspy-tools/getopt/LICENSE.md)

But after check libairspy/src is even more complex as it have a mix of MIT License & BSD 3-Clause License

nowls, 2018-04-10:
Hmm, yes that is more complex. What I'm looking for is a stable `LICENSE` or `COPYING` file whose contents won't change unless you decide to change the licensing under which your code is released. I believe this could be done like follows:

```
For file XXX:

<BSD 3-Clause license text>

For file YYY:

<MIT license text>
```
The Apache project does something like that: https://github.com/apache/httpd/blob/trunk/LICENSE

bvernoux, 2018-09-24:
In clear:
* airspy-tools examples are GPL in order to have contributions 
* LIB / DLL source code is MIT / BSD 3-Clause in order to integrate it in some software without redistributing or contributing ... (could be used in closed product)

nowls, 2020-03-04:
This was addressed satisfactorily.

closed 2020-03-04 by nowls

bvernoux, 2020-08-01:
Clarified in commit https://github.com/airspy/airspyone_host/commit/652fd7f1a8f85687641e0bd91f739694d7258ecc