airspy/airspyone_host #58: Build fails: because always_inline function '_mm_movehl_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'

yurivict, 2018-06-27:
On some systems clang-based buld fails:
```
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:470:23: error: always_inline function '_mm_set_ps' requires target feature 'mmx', but would be inlined into function 'translate_fs_4' that is compiled without support for 'mmx'
        ALIGNED __m128 rot = _mm_set_ps(hbc, 1.0f, -hbc, -1.0f);
                             ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:474:9: error: always_inline function '_mm_loadu_ps' requires target feature 'mmx', but would be inlined into function 'translate_fs_4' that is compiled without support for 'mmx'
                vec = _mm_loadu_ps(buf);
                      ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:475:9: error: always_inline function '_mm_mul_ps' requires target feature 'mmx', but would be inlined into function 'translate_fs_4' that is compiled without support for 'mmx'
                vec = _mm_mul_ps(vec, rot);
                      ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:476:3: error: always_inline function '_mm_storeu_ps' requires target feature 'mmx', but would be inlined into function 'translate_fs_4' that is compiled without support for 'mmx'
                _mm_storeu_ps(buf, vec);
                ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:123:15: error: always_inline function '_mm_set_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
        __m128 acc = _mm_set_ps(0, 0, 0, 0);
                     ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:139:19: error: always_inline function '_mm_loadu_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 head1 = _mm_loadu_ps(queue);
                                       ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:140:19: error: always_inline function '_mm_load_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 kern1 = _mm_load_ps(kernel);
                                       ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:141:19: error: always_inline function '_mm_loadu_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 head2 = _mm_loadu_ps(queue + 4);
                                       ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:142:19: error: always_inline function '_mm_load_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 kern2 = _mm_load_ps(kernel + 4);
                                       ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:144:18: error: always_inline function '_mm_mul_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 mul1 = _mm_mul_ps(kern1, head1);
                                      ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:145:18: error: always_inline function '_mm_mul_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        __m128 mul2 = _mm_mul_ps(kern2, head2);
                                      ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:147:11: error: always_inline function '_mm_add_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        mul1 = _mm_add_ps(mul1, mul2);
                               ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:149:10: error: always_inline function '_mm_add_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                        acc = _mm_add_ps(acc, mul1);
                              ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:181:17: error: always_inline function '_mm_loadu_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                __m128 head = _mm_loadu_ps(queue);
                              ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:182:17: error: always_inline function '_mm_load_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                __m128 kern = _mm_load_ps(kernel);
                              ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:183:16: error: always_inline function '_mm_mul_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                __m128 mul = _mm_mul_ps(kern, head);
                             ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:184:9: error: always_inline function '_mm_add_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
                acc = _mm_add_ps(acc, mul);
                      ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:202:13: error: always_inline function '_mm_add_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
        __m128 t = _mm_add_ps(acc, _mm_movehl_ps(acc, acc));
                   ^
/wrkdirs/usr/ports/comms/airspy/work/airspyone_host-1.0.9-18-gbfb6670/libairspy/src/iqconverter_float.c:202:29: error: always_inline function '_mm_movehl_ps' requires target feature 'mmx', but would be inlined into function 'process_fir_taps' that is compiled without support for 'mmx'
        __m128 t = _mm_add_ps(acc, _mm_movehl_ps(acc, acc));
                                   ^
```

bvernoux, 2018-06-28:
Could you push a pull request with a fix for that particular clang system (as of today we do not support clang and that requires some test ...) ?
Thanks to add more details on this clang system

bvernoux, 2019-04-07:
No answer / no details after few months this issue is closed

closed 2019-04-07 by bvernoux