airspy/airspyone_host #59: Can't find airspy

hemna, 2018-08-16:
I have a new airspy HF+ and plugged it in to my raspi 3 b+ running raspbian stretch

```
[ 1265.386636] usb 1-1.2: USB disconnect, device number 6
[ 2857.950265] usb 1-1.1.2: USB disconnect, device number 4
[ 2859.785473] usb 1-1.1.2: new high-speed USB device number 7 using dwc_otg
[ 2859.916146] usb 1-1.1.2: New USB device found, idVendor=03eb, idProduct=800c
[ 2859.916160] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2859.916168] usb 1-1.1.2: Product: AIRSPY HF+
[ 2859.916176] usb 1-1.1.2: Manufacturer: www.airspy.com
[ 2859.916186] usb 1-1.1.2: SerialNumber: AIRSPYHF SN:C852948094343FFC
```

You can see the kernel sees it, but running airspy_info can't find it.

```
root@airspy:~/openwebrx# airspy_info 
airspy_lib_version: 1.0.9
airspy_open() board 1 failed: AIRSPY_ERROR_NOT_FOUND (-5)
```

touil, 2018-08-16:
You are confusing Airspy One and Airspy HF+.

closed 2018-08-16 by touil

hemna, 2018-08-16:
I pulled down the repo for airspyhf, built it, and it doesn't have any of the airspy apps, such as
airspy_info, airspy_rx.

```
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/pkgconfig/libairspyhf.pc
-- Installing: /usr/local/lib/libairspyhf.so.1.4.1
-- Up-to-date: /usr/local/lib/libairspyhf.so.0
-- Up-to-date: /usr/local/lib/libairspyhf.so
-- Installing: /usr/local/lib/libairspyhf.a
-- Installing: /usr/local/include/libairspyhf/airspyhf.h
-- Installing: /usr/local/include/libairspyhf/iqbalancer.h
-- Installing: /usr/local/include/libairspyhf/airspyhf_commands.h
-- Installing: /etc/udev/rules.d/52-airspyhf.rules
```

touil, 2018-08-16:
Please do not use the issue tracker for discussing your projects. If you need to setup a websdr box, you can ask in the forums. For example: https://airspy.groups.io/g/main/messages