airspy/airspyone_host #66: Compilation issue in Windows 10 with MSVC/CMake

leonsal, 2019-12-03:
Hi,

When compiling for Windows 10 using MSVC (2019) and CMake, the compiler prints warnings
that it is ignoring an unrecognized option "-O3" for libairspy and "-O2" for airspy-tools.
But more importantly CMake is not passing the correct optimization options to MSVC (/O2 /Ob2),
so the resulting binaries are not optimized.

This issue can be solved by removing line:
`set(CMAKE_C_FLAGS_RELEASE "-O3")` 
from libairspy/CMakeLists.txt and
`set(CMAKE_C_FLAGS_RELEASE "-O2")`
from airspy-tools/CMakeLists.txt

`CMAKE_BUILD_TYPE=Release` already sets "-O3" for gcc/clang and sets "/O2 /Ob2" for msvc.
This implies, though, that airspy-tools will also be compiled with -O3 for gcc/clang.

Regards.

bvernoux, 2019-12-03:
Thanks for the feedback could you create a pull request with the modifications ? (like that you will contribute to the project)

bvernoux, 2019-12-04:
Fixed with https://github.com/airspy/airspyone_host/pull/67
Thanks for your contribution

closed 2019-12-04 by bvernoux