airspy/airspyone_host #82: Ship the udev rules upstream

AsciiWolf, 2021-11-04:
Please, consider submitting the [udev rules](https://github.com/airspy/airspyone_host/blob/master/airspy-tools/52-airspy.rules) to upstream (systemd). Here is an example how to do it: https://github.com/systemd/systemd/pull/20561

Without this, it is [not possible](https://github.com/flatpak/xdg-desktop-portal/issues/227#issuecomment-946760482) to use Airspy SDR devices in Flatpak and other special configurations when it is not possible to install the Airspy libs/udev rules system-wide.

bvernoux, 2021-11-05:
You are welcome if you want to do a pull request of the AirSpy udev rules to systemd and maintain it.

AsciiWolf, 2021-11-05:
It's not just Flatpak. AppImage apps, portable apps and basically all apps that cannot install custom things into the system /usr or /etc have the same problem. It is common on GNU/Linux that udev rules are shipped in upstream (udevd, which is now part of systemd). Missing udev rules are basically the only thing that prevents most of these devices from working out-of-box on Linux (airspy/rtl-sdr/etc. libs can be bundled as a part of the SDR app without being installed system-wide). I have [created](https://osmocom.org/issues/5291) the same ticket for rtl-sdr as well and will consider submitting the Airspy udev rules myself after the rtl-sdr ones are submitted.

bvernoux, 2022-02-25:
Could you provide an update about that issue ?

AsciiWolf, 2022-02-25:
Sorry, I sadly did not have enough free time to work on this. But if you want to start working on it and submitting the PR yourself, I will be glad to help.

bvernoux, 2022-02-25:
Thanks for your offer but I do not have time to work on that
So I will close this issue and if someone is interested to add that feature a Pull Request will be welcome

closed 2022-02-25 by bvernoux

AsciiWolf, 2025-05-13:
bladeRF was already added into systemd hwdb and can be used as an example of how the changes should look like:

https://github.com/systemd/systemd/blob/main/hwdb.d/70-software-radio.hwdb