airspy/airspyone_host #88: build fails on Raspberry Pi (debian buster)

miweber67, 2022-09-04:
Build fails due to `libusb_wrap_sys_device` not defined on this platform. 

```
pi@radiopi:~/radio/airspyone_host-master/build $ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
pi@radiopi:~/radio/airspyone_host-master/build $ cmake ../ -DINSTALL_UDEV_RULES=ON
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/radio/airspyone_host-master/build
pi@radiopi:~/radio/airspyone_host-master/build $ make
[ 41%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/airspy.c.o
/home/pi/radio/airspyone_host-master/libairspy/src/airspy.c: In function ‘airspy_open_device_fd’:
/home/pi/radio/airspyone_host-master/libairspy/src/airspy.c:757:11: warning: implicit declaration of function ‘libusb_wrap_sys_device’; did you mean ‘libusb_reset_device’? [-Wimplicit-function-declaration]
  result = libusb_wrap_sys_device(device->usb_context, (intptr_t)fd, &device->usb_device);
           ^~~~~~~~~~~~~~~~~~~~~~
           libusb_reset_device
[ 41%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/iqconverter_float.c.o
[ 41%] Building C object libairspy/src/CMakeFiles/airspy-static.dir/iqconverter_int16.c.o
[ 41%] Linking C static library libairspy.a
[ 41%] Built target airspy-static
[ 41%] Building C object libairspy/src/CMakeFiles/airspy.dir/airspy.c.o
/home/pi/radio/airspyone_host-master/libairspy/src/airspy.c: In function ‘airspy_open_device_fd’:
/home/pi/radio/airspyone_host-master/libairspy/src/airspy.c:757:11: warning: implicit declaration of function ‘libusb_wrap_sys_device’; did you mean ‘libusb_reset_device’? [-Wimplicit-function-declaration]
  result = libusb_wrap_sys_device(device->usb_context, (intptr_t)fd, &device->usb_device);
           ^~~~~~~~~~~~~~~~~~~~~~
           libusb_reset_device
[ 41%] Building C object libairspy/src/CMakeFiles/airspy.dir/iqconverter_float.c.o
[ 41%] Building C object libairspy/src/CMakeFiles/airspy.dir/iqconverter_int16.c.o
[ 41%] Linking C shared library libairspy.so
[ 41%] Built target airspy
[ 41%] Building C object airspy-tools/src/CMakeFiles/airspy_rx.dir/airspy_rx.c.o
[ 41%] Linking C executable airspy_rx
/usr/bin/ld: ../../libairspy/src/libairspy.so.1.0.11: undefined reference to `libusb_wrap_sys_device'
collect2: error: ld returned 1 exit status
make[2]: *** [airspy-tools/src/CMakeFiles/airspy_rx.dir/build.make:86: airspy-tools/src/airspy_rx] Error 1
make[1]: *** [CMakeFiles/Makefile2:249: airspy-tools/src/CMakeFiles/airspy_rx.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

miweber67, 2022-09-04:
Further digging reveals that the version of libusb-1.0 on the Pi is

`#define LIBUSB_API_VERSION 0x01000106`

and `libusb_wrap_sys_device` is not available until >= 0x01000107.

Not sure if the better path is trying to get a more recent version of libusb installed on the pi or to replace the call to `libusb_wrap_sys_device` with something equivalent available in the earlier version of libusb.

bvernoux, 2022-09-04:
Thanks for the issue and clear description
To be checked how to fix that with a clean fix

bvernoux, 2022-09-04:
Could you check if the fix https://github.com/airspy/airspyone_host/commit/64bc60b9ff27874abf7914d17937f4c898d056b6 solve your issue ? (it should)

miweber67, 2022-09-04:
Yes, 64bc60b fixes this problem! Tnx for quick turnaround. :-) We can close this issue.

Having some difficulty getting spyserver to attach to the device even though airspy_info works fine. Any tips or venue redirect to discuss?

bvernoux, 2022-09-04:
spyserver is not part of https://github.com/airspy/airspyone_host
If you find any issue related to airspyone_host components thanks to create an other issue

closed 2022-09-04 by bvernoux