airspy/airspyone_host #96: airspy_rx could use ppm correction option

zanco, 2024-02-29:
Hi, I am running airspy_rx for satellite strf recordings. The airspy_mini is on location, wired to 1545 MHz bandpassfilter. It's observations are 3 kHz higher as the satellites center frequency.  I can not run a calibrate tool cause I have no fixed / known signal source inside the bandpassfilter range.

In SDR software I can set + 2 ppm correction which makes the frequency match for my purpose.

However airspy_rx appears to have no option to add a ppm correction. Can this be added ? Thanks.

[image]

bvernoux, 2024-02-29:
For that purpose I have just created a new tool airspy_calibrate to read calibration data and erase/write calibration data in ppb (see for example https://www.sitime.com/ppm-hz-calculator)
See https://github.com/airspy/airspyone_host/commit/43570b63de90dacd2c5fc6500a538617fdb406ff

zanco, 2024-03-01:
Now running with -w 0 waiting for pass of SAR satellite. We will get there :-)

bvernoux, 2024-03-01:
> Now running with -w 0 waiting for pass of SAR satellite. We will get there :-)

If you really need something accurate to calibrate your AirSpy (or even to use the GPSDO on 10MHz External Reference Input on Airspy R2) I heavily recommend you to buy a GPSDO like https://www.leobodnar.com "Mini Precision GPS Reference Clock" which is quite cheap for excellent performances.

zanco, 2024-03-01:
[image]

It's spot on now ! Thanks a lot :-)

bvernoux, 2024-03-01:
Thanks for all so we can close that issue

closed 2024-03-01 by bvernoux