airspy/airspyone_host #99: vs2022 project doesnot work

intars-first, 2025-01-20:
These instructions:

[image]

in Visual studio 2022 lead to error:

error C1083: Cannot open include file: 'pthread.h': No such file or directory

bvernoux, 2025-01-21:
Do you have used that project https://github.com/airspy/airspyone_host/tree/master/libairspy/vc ? (it works with Visual Studio 2013 & 2019)

intars-first, 2025-01-21:
1. This time tried to download specifically vc folder (not using git as before)
2. Updated visual studio 2022 to include 2019 builder (or something)
3. Added additional source folders:
[image]
4. in the end it seems that #include <libusb.h> #include <pthread.h> referenced in airspy.c are missing.

bvernoux, 2025-01-21:
You shall follow the readme step by step libusb & libpthread shall be downloaded and extracted exactly like in documentation ...

closed 2025-01-21 by bvernoux